first95/FRC2019 #2: Develop approach to make Smart Dashboard not annoying

lindsayvallen, 2018-05-11:
We all hate having to drag all of the items from one corner to see things, so I think it's worth some time figuring out if there's a better way to use it.

jwalthour, 2018-09-20:
It might be worth looking at the other options in the drop-down on the drivers' station - I think there are like 3-4 alternatives to Smart Dashboard.  If any of them can be customized, they might do the trick!

lindsayvallen, 2018-10-19:
Documentation for SmartDashboard:
https://wpilib.screenstepslive.com/s/currentCS/m/smartdashboard
Also it includes this information on an advanced version of the dashboard that might be more stable:
https://wpilib.screenstepslive.com/s/currentCS/m/smartdashboard/c/92706

NHDinosaur, 2018-10-24:
The SFX SmartDashboard is no longer supported and was replaced by ShuffleBoard which is actually included in the driver station already.

NHDinosaur, 2018-11-07:
I developed a potential fix to the issue by uploading the save file smartdashboard uses to record the position of the widgets to github. Here are the instructions to make sure the fix is also on your computer:

In order to have the fix, you'll need to go to your SmartDashboard, go to file > preferences, click on the save path, and then go to your .git folder in the file selector to find the current save.xml file that has been uploaded. Once selected, you should have a save folder of the widgets that can't fail as easily.

NHDinosaur, 2018-11-14:
Did some further testing, and it seems to work as intended. As we fix other things on the robot, it would be great to see if it's still working this way, but for now I'm closing this issue.

closed 2018-11-14 by NHDinosaur