first95/FRC2019 #5: It would be great to be able to do sweeper turns

jwalthour, 2018-09-06:
It could be worth trying the Pathfinder library and or Motion Magic

jwalthour, 2018-11-29:
I started trying out the PathFinder library in the feature/add_pathfinder branch: https://github.com/first95/FRC2019/tree/feature/add_pathfinder

This branch includes the built .jar file version of PathFinder.

It seems like one of the major tradeoffs of this library is that it requires us to perform PID control of the drive motors in our code, on the RoboRIO, rather than on the Talons.  This implies that we'll also want to make sure that our code stays running on a regular schedule - we may need to try switching from the IterativeRobot base class to the TimedRobot base class.

NHDinosaur, 2019-10-02:
I'm not sure if autonomous mode will be brought back, but it's a good idea to fix this for 2020 just in case.