first95/FRC2019 #9: Determine which cameras to use for what

lindsayvallen, 2019-01-09:
We're going to use cameras in several ways this season: vision processing to identify hatch targets and floor targets, and possibly assisted driving during sandstorm. We should figure out which model camera we want to use for which functionality, or if there are other ones that we want to try. Outcome will be a(n initial) decision on the model to use for each functionality.

lindsayvallen, 2019-01-09:
We have a Microsoft LifeCam HD-3000 and a few Axis M1013s. Here are some useful references:

- Overall vision processing info from FRC 2019 https://wpilib.screenstepslive.com/s/currentCS/m/vision/c/86706
- How to configure Axis camera (at least one of our seems to already be configured) https://wpilib.screenstepslive.com/s/4485/m/24194/l/144985-configuring-an-axis-camera
- How to configure MS LifeCam https://wpilib.screenstepslive.com/s/currentCS/m/vision/l/288981-using-the-microsoft-lifecam-hd-3000
- Comparison discussion between Axis and LifeCam https://www.chiefdelphi.com/t/microsoft-lifecam-hd3000-v-axis-m1013-m1011-206-ethernet-camera/158670
- Comparison of Axis M1013 to prior M1011 and some of the quirks of M1013 https://wpilib.screenstepslive.com/s/currentCS/m/vision/l/288989-axis-m1013-camera-compatibility

lindsayvallen, 2019-01-09:
I connected an Axis M1013 and confirmed that is configured properly. Need to debug why NI Vision Assistant isn't installed properly, and then will try that out.

jwalthour, 2019-01-11:
James brings up a significant "pro" of the LifeCam: it's apparently significantly more mechanically durable.  He says the Axis cameras tend to shatter on minor impact, and that it shouldn't be a problem to mount a ring light around a LifeCam.

That sounds like a pretty strong argument in favor of the LifeCam to me.  Wanna just go with those?

lindsayvallen, 2019-01-19:
Main camera
- View vision targets on cargo ship, rocket ship, and loading station
- Provide general view of field for driving purposes during Sandstorm
- Provide general view of field for driving purposes during select other times during match, at drive team's request

Floor camera
- View vision targets on floor
- Maybe also provide general view of field for driving purposes

During Sandstorm, default camera settings will be for human visibility. Post-Sandstorm, default camera settings will be for vision processing.

closed 2019-01-19 by lindsayvallen