first95/FRC2019 #10: Calculate camera fields of view

lindsayvallen, 2019-01-12:
Calculate horizontal and vertical fields of view for LifeCam we've been using. See commit 4cff4877b85b9b627c48333b0b85c3f5be734dd9 for first pass at doing this for vertical field of view; will be similar for horizontal. Note that vertical should be re-done with two considerations:

- Target should be oriented along axis wish to calibrate, not off-axis (if do this, then modify script)
- Target board should be held in a plane parallel to the camera so that it's neither tilting toward or away from the camera.

These same considerations should be kept in mind for horizontal field of view too. This information will help us relate actual size, size in pixels in image, and distance away.

jwalthour, 2019-01-26:
This number also wants to be stored in this constant: `HatchVisionTargetsFromImage.CAMERA_FOV_WIDTH_DEG`

NHDinosaur, 2019-10-02:
Not sure if this will be relevant for 2020.

closed 2019-10-02 by NHDinosaur