first95/FRC2019 #13: Remove un-used components from 2019 code

lindsayvallen, 2019-01-19:
Specifically, remove all components, subsystems, and commands for the following: collector and climber. The robot drivebase and elevator will be similar enough this year to last year to keep and modify them.

lindsayvallen, 2019-01-19:
When done, make sure the code compiles. Then go through and check again that everything that should be removed has been, and everything that should stay hasn't been removed.

0NotApplicable0, 2019-01-25:
Removed all components, subsystems, and commands for the collector and climber. Also removed related controls and solenoid definitions in OI and Constants. Code builds on most recent WPI VSCode.

Do we also want to remove the random assignment for switch and scale as it is no longer going to be used?

jwalthour, 2019-01-25:
THanks, looks good!  Yeah, I think we can take those out.

0NotApplicable0, 2019-01-26:
All the sections for switch and scale operations were removed. NOTE: Automove selection was based of the switch and scale colors so it we try to run automoves getSelectedCommand only reutrns new Nothing(). Will need to fix how we select automoves.

closed 2019-01-26 by 0NotApplicable0

jwalthour, 2019-01-26:
Ah good point - thanks Nick!