first95/FRC2019 #16: NI and WPILib: get and install updates

lindsayvallen, 2019-01-19:
NI update available here (2019.1.0): http://www.ni.com/download/first-robotics-software-2015/5112/en/
Discussion of what this update includes: https://firstfrc.blob.core.windows.net/frc2019/Manual/TeamUpdates/TeamUpdates-combined.pdf

WPILib update (2019.2.1) available here: https://github.com/wpilibsuite/allwpilib/releases
Discussion of what this update includes: same as above

lindsayvallen, 2019-01-19:
I've completed it for CTC-17-ROBO-02. I did it without uninstalling the prior versions. I didn't notice any resulting issues during installation process, but I'll update later if I do.

jwalthour, 2019-01-25:
I've completed this on my laptop.

lindsayvallen, 2019-02-06:
Phil did those on his computer too so considering this closed.

closed 2019-02-06 by lindsayvallen