first95/FRC2019 #17: Determine interlocks among the sub-systems

lindsayvallen, 2019-01-19:
The sub-systems will not be fully independent of one another, e.g. when one sub-system is in a particular state, another sub-system may not be able to execute some of its functionality. To prevent the sub-systems interfering with or harming one another, we'll need to add interlocks. First, however, we'll need to determine more about the individual sub-systems and how they will mechanically interface to one another.

jwalthour, 2019-01-26:
So I don't forget it, here's one interlock: Shouldn't automatically approach a vision target with the elevator up

jwalthour, 2019-02-09:
When the climbing process is happening we don't want to be doing anything weapons-related.

jwalthour, 2019-02-16:
Semi-auto, semi interlock: when collecting cargo, the elevator should be slightly raised, such that the cargo collector is at about a 20 degree downward angle.  This removes collection dead zones and makes it less disastrous if the robot rams a wall with the collector down.

jwalthour, 2019-02-16:
Some of these are also recorded on the "interlocks" tab in this spreadsheet: https://docs.google.com/spreadsheets/d/17dYB1nawOt3gQoEPz9w_w4j_zVCD92162E-oP9PuYYQ/edit#gid=1600266100

jwalthour, 2019-02-17:
When skids are out, the hatch ground loader wrist should go up.  So should the cargo loader wrist, if possible.

jwalthour, 2019-02-17:
When elevator is at the bottom position and the hatch ground loader wrist is up, the hatch scorer should not extend or retract, as it will pass through the teeth of the pickup mechanism.

Conversely, it might be smart for the HGL wrist to not raise while the hatch scorer is extended, as this could cause the teeth to catch on the bottom of the retraction mechanism.

jwalthour, 2019-02-17:
When the hatch scorer is extended, it interferes with lowering the cargo loader wrist

jwalthour, 2019-02-20:
When the elevator is up high, robot should not shift into high gear

jwalthour, 2019-03-01:
When the hab climber pistons are extended, drivetrain should remain in low gear

NHDinosaur, 2019-10-02:
This is an interesting string of issues under one name. I know that the shift lock when the elevator is high has been implemented, but not sure on the others. Keeping for 2020.