first95/FRC2019 #18: Add live stream from camera via co-processor to Dashboard

lindsayvallen, 2019-01-19:


jwalthour, 2019-02-16:
For SmartDashboard, this must be done every time the dashboard is cleared.  These are the steps:

1. Enter editable mode (View->Editable or ctrl-E)
2. Click view->add->MJPEG stream viewer. A red rectangle will appear reading "NO CONNECTION".
3. Right-click the red rectangle and click "properties".
[image]
4. Double-click the cell to the right of "MJPG Server URL".  Paste in the address `http://frc95vision.local:1181/stream.mjpg`.
[image]
5. Press enter and close the properties box.  You will get a very tiny view.
[image]
6. Grab and drag the corners of the image to resize
[image]
7. Press ctrl-E again or click View->Editable to exit editing view

closed 2019-02-16 by jwalthour

jwalthour, 2019-02-17:
Now the pi is at the static IP address `10.0.95.60`, so the MJPG stream is now `http://frc95vision.local:1181/stream.mjpg`