first95/FRC2019 #23: Confirm which dashboard we want to use

lindsayvallen, 2019-01-22:
Main contenders are Smart Dashboard (what we've used before) and ShuffleBoard.

0NotApplicable0, 2019-01-28:
Last year we had issues with Smart Dashboard not updating the Choosers for auto moves, but if we are not going to use choosers and go with vision then sticking with Smart Dashboard might be a good idea. In the past we have had vision on the Smart Dashboard however at a point we had to reduce the quality because it was taking up too much bandwidth.

jwalthour, 2019-01-29:
This post has an interesting latency comparison: https://www.chiefdelphi.com/t/measuring-camera-display-latency/343508 

Unfortunately, they only compared the video streams on Google Chrome vs Shuffleboard, rather than comparing among the different dashboard options.

NHDinosaur, 2019-10-02:
We definitely want to use Smartdashboard, but the problem is that FIRST might stop supporting it in the near future. For 2020, I'll create a new issue explaining the same thing, and close this issue since we technically know which one we want to use.

closed 2019-10-02 by NHDinosaur