first95/FRC2019 #28: Automove Selection

0NotApplicable0, 2019-01-26:
How do we want to select automoves this year? A single mutable chooser that has simple and compound moves dynamically updated based off robot position?

Possibly three mutable choosers so drivers can create moves on the go?

jwalthour, 2019-01-26:
I just realized - with the [sandstorm](https://www.youtube.com/watch?v=gFsdPrBLHO0) we may not even need any auto move selection.

I think we should see how bad it is to drive the robot with the camera in first-person view, and if we need full automation, we'll want to come back to this issue.

0NotApplicable0, 2019-01-26:
Removed everything to do with auto move selection.

NHDinosaur, 2019-10-02:
This issue is just clogging space so I'm closing for 2020 repo.

closed 2019-10-02 by NHDinosaur