first95/FRC2019 #31: Do something about loop overrun errors

jwalthour, 2019-02-02:
For some reason, our main loop sometimes takes too long to run.  It's no longer keeping our robot from working, but it could cause us problems in the future.

Possible causes:

- We may have unnecessary CAN interactions.  For example, maybe we are applying PIDF coefficients from the smartdashboard every iteration.
- We should check for any loops doing a lot more things than necessary
- Are we doing any kind of data logging right now?
- Maybe we're sending too many items up to the smartdashboard?

jwalthour, 2019-02-06:
Someone from CTRE requested we collect some information regarding the issues we've been seeing.  We should collect this if/when an opportunity arises:
https://www.chiefdelphi.com/t/frc-95-the-grasshoppers-2019-build-thread/338781/139

If nothing else, we should save off and examine a log from the smartdashboard.

NHDinosaur, 2019-10-02:
This sounds like something we should fix for 2020