first95/FRC2019 #34: Update ground hatch loader subsystem

lindsayvallen, 2019-02-02:
Ground hatch loader will have:
- One motor/encoder for rollers to pull the hatch in
- One motor/encoder for wrist to rotate hatch to vertical orientation
Currently, our stubbed ground hatch loader has:
- Two motors/encoders for rollers to pull the hatch in
- One pneumatic for wrist to rotate hatch to vertical orientation

jwalthour, 2019-02-06:
Sorry, I missed the feature branch, and made a quick-and-dirty version to prepare for hardware testing.  My quick-and-dirty version is in 562b8e13930e907f5d8891797fbc2f18f058925d, but I think your more thought-out version is the one we'll want to use in the end.

lindsayvallen, 2019-02-06:
@jwalthour That's okay. I made a lovely merge of your develop version into my branch and I'm moving on from there :)

lindsayvallen, 2019-02-06:
From discussion with James, the wrist should be implemented as follows:
- Wrist should have two primary positions, up and down, with buttons to select between those positions.
- When the wrist is up, its motor will need to be active to maintain that position.
- When the wrist is down, the motor should be inactive so that the wrist has some give to bounce around a bit without being driven down.

My interpretation of these desires is that we should have wrist up and wrist down buttons, and the wrist will need control similar to the elevator to achieve and maintain positions.

lindsayvallen, 2019-02-06:
@jwalthour After merging your version and my version of the HGL and making a few small tweaks to prepare it for future position control, I've merged the feature branch back into develop. That way if we need to continue using this basic, speed-based HGL code, it's available and more consistent with the feature branch. In the meantime, I'll continue developing the position control of the HGL wrist in the feature branch because it will be significant changes -- making it more like the elevator, able to hold a position.

lindsayvallen, 2019-02-08:
Request from Andy: look for current spike from HGL intake too to determine when it's grabbed a hatch; that can then be used in semi-auto move to have wrist pop back up (part way or all the way) after a hatch is successfully loaded.

lindsayvallen, 2019-02-09:
For the compound auto-collect command, here's a list of sub-commands:
- Turn on the intake to pull in (rather than push out)
- Lower the wrist to the ground
- Detect current spike in intake motor, and stop intake motor
- Raise wrist to its vertical position (or perhaps half-way up to avoid some interlock issues?)
- Transmit rumble to weapons controller?

jwalthour, 2019-02-12:
Compound auto-collect command is now `commands/hgroundloader/AutoAcquire.java`, as of f84f32d861b30aec621420fe09b56d31c4a75c6f

lindsayvallen, 2019-02-14:
Tried auto acquire command with some necessary updates per commit 02b9cbe. Started the intake, had the wrist down, but didn't ever stop intake even after got hatch. Need to plot the current for the HGL intake and adjust the current limit params accordingly.

lindsayvallen, 2019-02-17:
In feature/better_hgl the hatch ground loader is working in the same kind of way as the cargo handler and in the same way. However, the AutoAcquire move doesn't work yet. I think that needs to get incorporated into the CombinedControlHGroundLoader command because the latter needs to be active to maintain wrist position, but the former needs to interrupt it. @jwalthour is there a way for AutoAcquire to interrupt the default command, and then when done, go back to the default command?

jwalthour, 2019-02-18:
I'm not sure, but I think the interruption behavior you describe is the default scheduler behavior.  It might be necessary for the default command to set itself to be interruptible (i think there's a function called setInterruptible(true) or something like that).

NHDinosaur, 2019-02-19:
There is a function called setInterruptible(true), but I'm not quite sure how to implement it. I did add it to CombinedControlHGroundLoader and put it at the end of the Start command, since we would only need to set it once.

closed 2019-03-01 by jwalthour