first95/FRC2019 #35: Compare XBox One controller to the XBox 360 controllers that we have

lindsayvallen, 2019-02-06:
We have XBox 360 controllers that we have used, but no spares and we don't seem to be able to buy those any more. Instead, we have ordered an XBox One controller (scheduled to arrive 2/7) and want to see how it compares to the 360 to decide whether to buy more of those or look for second-hand 360s. The main thing to check is whether the One has similar button/axis mappings, e.g. at least as many of each, and whether there are any other issues with it.

nparnold, 2019-02-07:
As for the physical model of an Xbox One controller to an Xbox 360 controller (I own both) they are exactly the same and I am pretty sure that they would have the same software mappings as well.

lindsayvallen, 2019-02-08:
We got an XBox One, and using Driver Station, I compared that it has the same buttons and button mapping as the XBox 360 and seems similar. The WPILIB XBoxController class works for both controllers (http://first.wpi.edu/FRC/roborio/beta/docs/java/edu/wpi/first/wpilibj/XboxController.html) and seems to work the same way. Waiting to try it out a bit with driving, and if that's okay too, then we can close this issue and buy more XBox One controllers.

lindsayvallen, 2019-02-08:
Used XBox One controller to operate the HGL (y axis and triggers) and HS (bumpers) and it seems to work basically the same as the XBox 360. Given that the 360s are hard to find and the One's aren't very expensive ($60) and we at least need another back-up for the 360s, we should get at least one more One controller now. Then we can use those controllers for a while, and decide if we want more in the future.

closed 2019-02-08 by lindsayvallen