first95/FRC2019 #54: Rumble commands only work with the driver controller

jwalthour, 2019-03-01:
At first I thought this was a hardware issue, but I also saw this problem with a controller that I'd confirmed was working.

jwalthour, 2019-03-01:
Might also make sense to rework this into just using "left" and "right" rumbler constants rather than bothering with "high pitch" and "low pitch".  Both Phil and Andy perceive it as "right" and "left" rather than varying pitches, so this might just be a fluke in my perception priorities.

lindsayvallen, 2019-03-20:
Would like to rumble both controllers, or at least with weapons controller, preferably before 3rd event.

jwalthour, 2019-04-06:
It appears that the driver's station is incapable of making the weapons controller rumble, manually or from the code: 

https://photos.app.goo.gl/hAgz4gxgt4RsmSGu6

jwalthour, 2019-04-06:
Fixed (for the driver controller) in 16407b566bab603745d42ccf1ecc52e060b016fb .  The weapons controller bit seems like a driver's station bug.

closed 2019-04-06 by jwalthour