first95/FRC2019 #60: Add autonomous routine to navigate the robot towards a target based on a white line on the floor

jwalthour, 2019-03-02:
Depends on #59

lindsayvallen, 2019-03-20:
Implemented this in AutoSteerAlongLine command, and verified that it does work during practice at event 2. However, need to make sure it has sufficient parameters to calibrate/tune it and do so. Specifically, at least need forward and turn rates based on sensors' values, maybe some others too.

lindsayvallen, 2019-03-20:
Also, incorporate two additional front-facing sensors that will trigger based on whether something (i.e. wall of loading station) is in front of them. Use these to give confirmatory input about whether lined up and know when close enough. When close enough, then want to rumble controllers (depends on issue #54 ).

lindsayvallen, 2019-03-21:
Need to think about logic handling for checking the sensor values. Currently, if middle sensor is true then we don't check the others so if line is detected by more than one sensor (and hence robot is rotated a bit from desired), we won't notice and turn. So maybe want to check sensors from outside in rather than inside out. Or check multiple sensors. @jwalthour You wrote this originally and seemed to go outside in intentionally; what do you think?

jwalthour, 2019-03-21:
My thinking with going inside-to-outside was to interpret the sensor input in the way that would result in the least change in direction.  That way if we get minor noise in the readings, the robot's reaction isn't sudden and dramatic.

My thinking with making it respond only to one sensor's input was to make for a trivial way to avoid strange scenarios.  For example, we could theoretically get input like 1 0 0 0 1 1 1 (from left to right, where 1 is a sensor with a detection).  Checking one sensor at a time is a simple way to remain tolerant of all strange inputs.

But now that I think of it, some kind of weighted average might be an even better way to do it.  For example, we could assign a turn rate to each sensor.  That'd give us some vector like [-1 -0.5 -0.25 0 +0.25 +0.5 +1]  So for an input like [1 0 0 0 1 1 1], the turn command would be (-1 + 0 + 0 + 0 + 0.25 + 0.5 + 1)/4.

lindsayvallen, 2019-03-26:
Switched sign of forward speed calculation because it was going backwards, tested it without the forward sensors and it worked well enough that I merged it into develop. We still need to add and test the forward-facing sensors, and that should probably still be done in the branch.

lindsayvallen, 2019-03-28:
Integrated forward sensors so that autosteer command stops when both forward sensors are activated, and then rumbles driver controller. Not ideal, but works okay. Merged into develop.

closed 2019-03-28 by lindsayvallen